miketeachman/micropython-i2s-examples #4: Example scripts: Pin names are platform specific

peterhinch, 2021-08-11:
This occurs in various scripts, for example `play-wav-from-flash-blocking.py`. The script includes platform detection code which checks for STM and ESP32, but then goes on to define pins in a way which is not STM-compatible.
```python
SCK_PIN = 33
WS_PIN = 25
SD_PIN = 32
I2S_ID = 1
```
I appreciate this is obvious to experienced users, but these scripts are likely to be run by those new to MP.

miketeachman, 2021-08-11:
That's a good point.  I can see how this would be confusing.  I can change the scripts to indicate that a port-specific pin designation is needed, for example:

```
SCK_PIN = <enter SCK pin designation>
WS_PIN = <enter WS pin designation>
SD_PIN = <enter SD pin designation>
I2S_ID =<enter I2S peripheral ID designation>
```
The scripts could also include comments pointing users to the port-specific pin and ID possibilities, eg:
https://github.com/miketeachman/micropython-i2s-examples#pyboard-gpio-mappings-for-sck-ws-sd

What do you think would work best to lead new users down the right path?

peterhinch, 2021-08-13:
The approach I use is to detect the platform at runtime (as you are doing) and assign appropriate pins accordingly. Then document which platforms work "out of the box" and which pins to use. Crudely (you might want to extend to other platforms):
```python
_stm = True
if uos.uname().machine.find("PYBv1") == 0:
    pass
elif uos.uname().machine.find("PYBD") == 0:
    import pyb
    pyb.Pin("EN_3V3").on()  # provide 3.3V on 3V3 output pin
elif uos.uname().machine.find("ESP32") == 0:
    _stm = False
else:
    print("Warning: program not tested with this board")  # +stm = ???

# ======= AUDIO CONFIGURATION =======
WAV_FILE = "side-to-side-8k-16bits-stereo.wav"
WAV_SAMPLE_SIZE_IN_BITS = 16
FORMAT = I2S.STEREO
SAMPLE_RATE_IN_HZ = 8000
# ======= AUDIO CONFIGURATION =======

# ======= I2S CONFIGURATION =======
if _stm:
    SCK_PIN = "Y9"
    WS_PIN = "Y4"
    SD_PIN = "X22"
else:
    SCK_PIN = 33
    WS_PIN = 25
    SD_PIN = 32
I2S_ID = 1
```

miketeachman, 2021-08-14:
Hopefully this lastest commit https://github.com/miketeachman/micropython-i2s-examples/commit/bafeb0ea213bcc4868d320b181bd3383f7510a00 makes the pin assignments more obvious.

peterhinch, 2021-08-15:
Great improvement, and to the docs. Easily run demos will save you a lot of hassle in the future :)

closed 2021-08-15 by peterhinch