miketeachman/micropython-i2s-examples #6: format PDM or PCM?

FernandoQui, 2021-08-13:
Hi. Great job in api machine i2s. 
If I want to send audio data from a microphone using the API_i2s  through DAC of an ST board, should I filter and decimate the data buffer output gotten from API_i2s to convert it  in PCM data, or should I just take the data output and send it to DAC's buffer ?
Does the API_i2s  transform from PDM to PCM or does it just work with PDM?  
What kind of modulation is recording in the .py examples scripts: PCM or PDM?

miketeachman, 2021-08-15:
Hi,
I'm having a bit of difficulty to understand your application.  Is it possible to describe a bit more on the type of microphone you have (PDM or I2S?) and how you would like to use it?

thanks!

> Does the API_i2s transform from PDM to PCM or does it just work with PDM?

The I2S Module only works with PCM data

> What kind of modulation is recording in the .py examples scripts: PCM or PDM?

PCM

FernandoQui, 2021-08-18:
Hello, thanks for responding so soon. I will explain in more detail the project I am working on. I'm using the ST microphone: MP45DT02 (with I2S interface, data output in PDM format); the microcontroller is: ST32F407-ARM-CortexM4.I'm using the last compilation of Micropython, to access I2S chanel and the device.
Basically I am doing a test reading the audio data on the microphone and sending it to the DAC that the onboard microcontroller has. 
From the answer you gave me above, and reviewing in more detail the hardware on which the tests were made, the API receives and delivers the data directly in PCM format, since the hardware of the audio amplifier and microphone are of this type of modulation.
I will have to do filtering and decimation processing to get results. If you are interested we could work together to extend the library to support PDM modulated microphones.

miketeachman, 2021-08-19:
Hi,
Thank you for describing the application.  I took a look at the datasheet.  Unfortunately it will not be possible to use this PDM microphone with the new I2S class.  Sorry.  These two protocols are not compatible.   Sometime in the future a new PDM class could be added to the Machine module of MicroPython to support PDM microphones.  Maybe you would like to take a look at that work?

But, some good news, the CircuitPython documentation shows that PDM microphones are supported:
https://learn.adafruit.com/adafruit-pdm-microphone-breakout/circuitpython

Is it important to use a PDM microphone?  If you switched to an I2S microphone you can use this new I2S class with no changes.

miketeachman, 2022-02-21:
There are no plans to support PDM microphones in the machine.I2S class, so closing this issue.

closed 2022-02-21 by miketeachman