miketeachman/micropython-i2s-examples #7: Is there a way to cancel playback?

peterhinch, 2021-09-10:
I'm hoping to do it without calling **.deinit()** so that the interface remains active. The following code stops playing if another task sets `self.song=None` but it does not stop cleanly, issuing a continuous tone.
```python
    async def play(self):
        swriter = self.swriter
        wav_samples_mv = memoryview(wav_samples)
        with open(self.song, "rb") as wav:
            _ = wav.seek(44)  # advance to first byte of Data section in WAV file
            while (num_read := wav.readinto(wav_samples_mv)) and self.song is not None:
                I2S.shift(buf=wav_samples_mv[:num_read], bits=16, shift=self.volume)
                swriter.write(wav_samples_mv[:num_read])
                await swriter.drain()
        self.song = None
```
I'm guessing there's a way to issue `swriter.write(silence)`. What is the minimum which would constitute silence?

As an aside you might be interested to hear that asynchronous I2S playback on a Pyboard D SF2W works with my micro-gui library. This with 44.1KHz 16 bit audio and attendant buffers. With the display I'm using the GUI repeatedly blocks for ~40ms.

miketeachman, 2021-09-13:
> In practice filling `wav_samples` (currently 15KiB) with zeros and outputting that fixes the problem, but I'd like to understand why this is necessary. As far as I can see the chip (UDA1334) datasheet sheds no light on this

Here's what's happening:  the design is presently based on the assumption that a continuous stream of audio samples will be written into the I2S module.  When this stream stops and an underflow condition happens, the I2S module continues to play the last block of audio samples (256 bytes), which is what you hear as a tone.  When samples with zero value are written into the I2S module last block the non-zero audio samples are flushed out.

It's possible to change this behaviour by detecting the underflow condition and writing zeros into the internal peripheral interface.  That should solve the problem you see ... or rather hear :)   Please let me know what you think.

> As an aside you might be interested to hear that asynchronous I2S playback on a Pyboard D SF2W works with my micro-gui library. This with 44.1KHz 16 bit audio and attendant buffers

Fantastic !   I've done some limited I2S stress tests using asyncio and have found that the Pyboard D is the most performant of all the ports (better than the ESP32 and Pyboard v1.1).   I'm guessing that it performs better than the ESP32 because it's not burdened by an underlying RTOS and I've read that it has an efficient interrupt handling system.

peterhinch, 2021-09-13:
In terms of practical application code you need a buffer to store data from the disk before yielding to uasyncio. The size of that buffer is easy to calculate if you know the uasyncio latency of your application and your audio sample rate. In my case 15KiB works out right, and I pre-allocate that. So the buffer exists, filling it with zeros and writing it out is trivial. Users of the other drivers will need to send 256 bytes of zeros. I think documenting this behaviour might be sufficient, but I have no strong feelings on it. I was just puzzled as to what was going on.

In my testing 40000 seems to be the required I2S buffer size. It would be interesting to know how this is calculated. You certainly can't reduce it.

I'm not surprised by the difference between Pyboard and ESP32. The ESP32 interrupt latency is diabolical and it only supports soft IRQ's. Latency if SPIRAM is fitted is 100ms! That is how long a GC takes, checking all that RAM. I regard it as pretty useless for performance unless you can use native hardware like the RMT device. I've not yet studied or tried your I2S on ESP32 but I assume it uses native hardware. I wonder if you've tested it on a board with SPIRAM. I have my doubts whether a practical application would be possible. On the one hand you've got plenty of RAM for buffers, but on the other hand you won't half need it...

I am surprised you see a significant difference between Pyboard D and Pyboard 1.1. Aside from the obvious differences in resources both have very low Python interrupt latency - on the order of 15μs. However you've gone in much deeper than me - I have no experience with coding DMA on any MicroPython hardware.

I'm looking forward to seeing your Pi Pico version. I'd imagine that using the PIO should be straightforward. It's an amazingly cost-effective piece of kit. Apart from maths performance it's fast and the PIO is wonderful.

peterhinch, 2021-09-14:
I added more pushbuttons to my playback application and ran into RAM issues (on a Pyboard D SF2W). Currently I'm not using frozen bytecode so the entire GUI is running in RAM. I'm confident freezing code will fix this, but the thing was a proof of concept to check for realtime issues so I'm not sure I'll pursue it further.

The bottom line is that I believe a hi-fi quality media playback app could be built using my GUI, with the usual play-stop-pause-skip capability and the ability to choose from a list of directories containing albums. But frozen bytecode or an SF6W would be needed. A Pyboard 1.1 might present a challenge.

miketeachman, 2021-09-15:
> In my testing 40000 seems to be the required I2S buffer size. It would be interesting to know how this is calculated. You certainly can't reduce it.

The size of the internal buffer is related to the max amount of time that other tasks can run before the i2s write method is allowed to run by the scheduler.  For example, assume a 40kB internal buffer and a WAV file with 44.1kHz sample rate, 16-bit, and stereo.  40000 bytes/44100/(16/8)/2 = 22.7 ms.  If the i2s write method is blocked from running more than 22.7 ms, then the internal buffer will be emptied by the DMA and underflow will happen, leading to gaps in the audio output.

> I am surprised you see a significant difference between Pyboard D and Pyboard 1.1

The Pyboard v1.1 has about 1/2 the DMIPS as the Pyboard D  -- I figured that explained the performance difference.

> I'm looking forward to seeing your Pi Pico version.

It's slowly getting done - I can play a 16-bit WAV file using the UDA1334.  Lots of hard-coding and not much error checking.  I haven't started on the readinto() method.

> The bottom line is that I believe a hi-fi quality media playback app could be built using my GUI

Very cool !

peterhinch, 2021-09-15:
I have to query your maths which seems to be out by a factor of 10:
```python
>>> 40000/44100/(16/8)/2
0.2267573696145125
```
i.e. 226ms. I find multiple level division confusing, but doing the arithmetic my way produced the same result :) Yet I found 40,000 to be empirically necessary. 30,000 did indeed produce gaps.

My disk buffer size is 15KiB which should handle up to 87ms of blocking:
```python
>>> 15*1024/(44100*2*2)
0.08707482993197278
```
This seems to accord with empirical testing, so I was puzzled that the DMA buffer needed to be so much bigger.

miketeachman, 2021-09-15:
doh!  thanks for catching that calculation mistake.

> This seems to accord with empirical testing, so I was puzzled that the DMA buffer needed to be so much bigger.

I'd like to understand this as well.  What display are you using with the micro-gui library?  I'd like to try it out the library with the display you're using and then add timing instrumentation (print and printf calls) to track down the root cause of this situation.

miketeachman, 2021-09-16:
I integrated the nano-gui scale.py demo into the [uasyncio example](https://github.com/miketeachman/micropython-i2s-examples/blob/master/examples/play_wav_from_sdcard_uasyncio.py) in this lib.   For hardware I used an ESP32 SPIRAM and an ili9341 display.   Ticks_us()/print instrumentation was added.  The results some light on why a larger than expected ibuf is needed, at least for the ESP32.  The essential finding is this:   the I2S object needs to buffer enough samples to be able to ride through the times when the nano-gui routines are being called by the scheduler.  I guess that is no surprise.  In the scale.py demo, the 2 calls to scale.value() take about 40ms and 65ms, respectively.  Each segment inside the do_refresh() call takes about 50ms (and there are 4 per each display update).    When the nano-gui routines are sleeping, the I2S object is filling the internal buffer.  If the I2S internal buffer is too small it will completely empty when the nano-gui routines are running -- this leads to underflow and the broken audio stream.

I'd still like to do some investigation into the Pyboard D behaviour.  The STM32 port is a different implementation, but I think the results will be the same.   There may still be other undiscovered factors that require a large ibuf.

Note:  It's been fun seeing what the nano-gui library can do -- it's is quite an impressive piece of cross-platform engineering.

peterhinch, 2021-09-16:
The nano-gui **scale** demo is quite dynamic and demanding on uasyncio.

I did my testing with [micro-gui](https://github.com/peterhinch/micropython-micro-gui) because this allows user input, with code based on [simple.py](https://github.com/peterhinch/micropython-micro-gui/blob/main/gui/demos/simple.py) which is very minimal. I just had a single on-screen pushbutton which started playback. The main load on uasyncio is the display update segments (it uses the same driver as nano-gui), but there are a number of other coros checking pushbuttons and suchlike. These only block for short times. My display is a 1.5" SSD1351 128x128 pixel and the host is a Pyboard D SF2W.

The nub of my question is, when my uasyncio loading is such that I can run with a disk buffer of 16KiB, why do I need a DMA buffer of 40,000 bytes? On the face of it I'd expect the same timing constraints to apply to both.

This is the disk reading code (based on yours):
```python
    async def play(self):
        swriter = self.swriter
        wav_samples_mv = memoryview(wav_samples)  # Global 16KiB bytearray
        with open(self.song, "rb") as wav:
            _ = wav.seek(44)  # advance to first byte of Data section in WAV file
            while (num_read := wav.readinto(wav_samples_mv)) and self.song is not None:
                I2S.shift(buf=wav_samples_mv[:num_read], bits=16, shift=self.volume)
                swriter.write(wav_samples_mv[:num_read])
                await swriter.drain()
        for x in range(len(wav_samples)):  # Neccessary for silence
            wav_samples[x] = 0
        swriter.write(wav_samples)
        await swriter.drain()
        self.song = None
```

miketeachman, 2021-09-16:
Can you share the full code for your test?  I'd like to reproduce your results.  A ST7735 display is the closest I have to a SSD1351 display.   The driver code for the ST7735 reads `# Blocks 38.6ms on Pyboard D at stock frequency`.  You mentioned earlier in this thread that the SSD1351 display update blocks for ~40ms.  Based solely on the similarity of update times, it looks like a ST7735 display will be a reasonable proxy for a SSD1351 display.  I'll test with a Pyboard D.

peterhinch, 2021-09-16:
When I first tested this I couldn't reduce the DMA buffer below 40,000 without glitches. However I've tried again and can't replicate this so maybe I was doing something daft. It seems to be working with both buffers set to 20KiB. But you're welcome to experiment.

The code below is stripped down as my demo has some special resources and was running into memory problems (with a 40000 byte buffer). This will run against standard [micro-gui](https://github.com/peterhinch/micropython-micro-gui). As I'm sure you've figured out, you'll need to adapt `hardware_setup.py`, supply at least two pushbuttons, and provide a music file.

```python
import hardware_setup  # Create a display instance
from gui.core.ugui import Screen, ssd

from gui.widgets.buttons import Button, CloseButton
from gui.widgets.sliders import HorizSlider
from gui.core.writer import CWriter

# Font for CWriter
import gui.fonts.arial10 as arial10
from gui.core.colors import *

import os
import uasyncio as asyncio
from machine import I2S
from machine import Pin
import pyb

pyb.Pin("EN_3V3").on()  # provide 3.3V on 3V3 output pin

# ======= I2S CONFIGURATION =======

I2S_ID = 1
BUFSIZE = 1024*20
# allocate sample array once
wav_samples = bytearray(BUFSIZE)  # 5.8ms/KiB

config = {
    'sck' : Pin('W29'),
    'ws' : Pin('W16'),
    'sd' : Pin('Y4'),
    'mode' : I2S.TX,
    'bits' : 16,  # Sample size in bits/channel
    'format' : I2S.STEREO,
    'rate' : 44100,  # Sample rate in Hz
    'ibuf' : BUFSIZE,  # Buffer size
    }

# ======= GUI =======

class BaseScreen(Screen):

    def __init__(self):
        args = {
                'bdcolor' : RED,
                'slotcolor' : BLUE,
                'legends' : ('-48dB', '-24dB', '0dB'), 
                'value' : 0.5,
                'height' : 15,
                }
        super().__init__()
        self.song = None
        self.volume = -3
        self.reg_task(asyncio.create_task(self.main()))
        # verbose default indicates if fast rendering is enabled
        wri = CWriter(ssd, arial10, GREEN, BLACK, verbose=False)
        col = 2
        row = 2
        Button(wri, row, col, text='Play', callback=self.play_cb, fgcolor=GREEN)
        row = 30
        Button(wri, row, col, text='Stop', callback=self.stop_cb, fgcolor=GREEN)

        row = 110
        col = 14
        HorizSlider(wri, row, col, callback=self.slider_cb, **args)
        CloseButton(wri)  # Quit the application

    def slider_cb(self, s):
        self.volume = round(8 * (s.value() - 1))

    def play_cb(self, *_):
        self.song = "/sd/music/spies.wav"

    def stop_cb(self, *_):
        self.song = None

    # Open and play a binary wav file
    async def play(self):
        swriter = self.swriter
        wav_samples_mv = memoryview(wav_samples)
        with open(self.song, "rb") as wav:
            _ = wav.seek(44)  # advance to first byte of Data section in WAV file
            while (num_read := wav.readinto(wav_samples_mv)) and self.song is not None:
                I2S.shift(buf=wav_samples_mv[:num_read], bits=16, shift=self.volume)
                swriter.write(wav_samples_mv[:num_read])
                await swriter.drain()
        for x in range(256):  # Neccessary for silence
            wav_samples[x] = 0
        swriter.write(wav_samples_mv[:256])
        await swriter.drain()
        self.song = None

    async def main(self):
        audio_out = I2S(I2S_ID, **config)
        self.swriter = asyncio.StreamWriter(audio_out)
        try:
            while True:
                if self.song is not None:
                    await self.play()
                else:
                    await asyncio.sleep_ms(200)
        finally:
            # cleanup
            audio_out.deinit()
            print("==========  CLOSE AUDIO ==========")

def test():
    print('Audio demo.')
    Screen.change(BaseScreen)  # A class is passed here, not an instance.

test()
```

peterhinch, 2021-09-17:
Perhaps you could let me know how you'd like to proceed. I plan to produce a demo of a player with the usual play-pause-stop functionality. I'm confident I can get this working on an SF2. There is at least one realtime issue around cancellation, but thanks to your help I think I understand it and know how to deal with it.

One option is for me to complete writing the demo then let you have the code and implementation notes.

On the other hand if you want to experiment with this now, I could create a branch of micro-gui with the code in its half-baked state plus resources and we could maintain a dialog on the details.

You might be interested in another thing I'm working on: a way of monitoring uasyncio applications. The idea is to show, on a scope or logic analyser, when tasks start and stop. In the code under development you use a decorator to identify coros you want to monitor. You hook up the system under test to a Pi Pico using a single GPIO (currently a UART Tx), and attach your LA to the Pico. Blocking can be identified by monitoring a null coro: the duration between consecutive runs can be measured and any long gaps matched against the coros which are running at the time.

As a general query, do you happen to know if hardware exists with I2S interface and Toslink or S/PDIF output? It seems an obvious thing to offer but I haven't spotted anything so far.

miketeachman, 2021-09-18:
> The code below is stripped down as my demo has some special resources and was running into memory problems (with a 40000 byte buffer). This will run against standard [...]

Thanks a ton for making this example.  I'll experiment with different sizes of application and internal buffer sizes.  No doubt I'll learn more about the I2S performance with uasyncio.  I'll have time tomorrow to work on this.

miketeachman, 2021-09-18:
> One option is for me to complete writing the demo then let you have the code and implementation notes.
> 
> On the other hand if you want to experiment with this now, I could create a branch of micro-gui with the code in its half-baked state plus resources and we could maintain a dialog on the details.

I'd like to help contribute to the player, but, at the moment, I'm trying to focus on the I2S port to the Pico.  Unfortunately, I never find as much time as I'd like for open source projects.  In a couple of weeks I should have the Pico I2S PR finished.  I'll have more time then.  A wav player would make an excellent addition to your uasyncio tutorials

> The idea is to show, on a scope or logic analyser, when tasks start and stop

That would be a game changer for uasyncio development !   A visual indication of coro operation would be huge.   It would really help for IO related work, to see what the scheduler is doing.  Also,  with the scope I have (the ubiquitous hobbyist DS1054) it would possible to set triggers on missed deadlines, or coros running too long.  That would enable long term tests to catch hard to detect timing issues.

> As a general query, do you happen to know if hardware exists with I2S interface and Toslink or S/PDIF output? It seems an obvious thing to offer but I haven't spotted anything so far.

No, sorry.  I had to google both those standards.

peterhinch, 2021-09-18:
> I'm trying to focus on the I2S port to the Pico.

I'm sure that's the best use of your time. I'll let you have details of the player demo and the uasyncio monitor when they are in a reasonable state, but aside from that I'll let you get on.

There is one issue regarding cancellation which you might want to think about in the background. I'm confident I can work around it, but you may want to consider it.

To cancel playback it is necessary to let the current file buffer to empty, then zero the first 256 bytes and send that. For example this approach works. The `.playing` variable can be cleared by other tasks to pause playback.
```python
        with open(self.song, "rb") as wav:
            _ = wav.seek(self.offset)
            while (num_read := wav.readinto(wav_samples_mv)) and self.playing:  # .playing causes a synchronous end
                I2S.shift(buf=wav_samples_mv[:num_read], bits=16, shift=self.volume)
                swriter.write(wav_samples_mv[:num_read])
                await swriter.drain()
                self.offset += size
        # When it gets here it's ready for a new buffer
        for x in range(256):  # Necessary for silence
            wav_samples[x] = 0
        swriter.write(wav_samples_mv[:256])
        await swriter.drain()
```
The apparent solution of cancelling the task and trapping `asyncio.CancelledError` is tricky. I think this is because the cancellation can occur while `await swriter.drain()` is in progress. In any event I haven't yet managed to get that approach to work cleanly.

I think cancellation of playback is important. It can be made to work with care. Whether use could be simplified with an addition to the API is one I'll leave to you to consider.

miketeachman, 2021-09-19:
> I think cancellation of playback is important. It can be made to work with care. Whether use could be simplified with an addition to the API is one I'll leave to you to consider.

It's definitely possible to extend the API to make it easier.  The I2S class is in a "technology preview" phase where the API is adaptable based on feedback.  There are two issues I'm looking at:
1.  defaulting to silence when underruns happen (rather than continuing to send out the samples remaining a buffer).  That can be done for all platforms.  The ESP-IDF actually has an API config for exactly that purpose.  The STM32 and rp2 changes are straightforward as the relevant buffers are not hiding behind a manufacturer API.

2. stopping playback immediately.  Don't send out the residual samples in the internal buffer.  Is that the behaviour that is preferred?   If so, I think it's doable.  Perhaps new methods:

- `I2S.stop()`   immediately stop sending samples via DMA and clear the internal buffer.  Ignore write() and readinto() methods.
- `I2S.resume()`  resume normal operation

Does this fit with your anticipated use pattern?

Here's something that you might consider useful for a music player - a wav header parser to pick out the attributes needed for playback.  It's integrated into one of the [helper py files](https://github.com/miketeachman/micropython-i2s-examples/blob/master/examples/wavplayer.py) used in an example.

For the rp2 work I'm doing now, one major thorn I'm hitting is the SD card support.  At the moment there is no machine.SDCARD for the rp2, just the sdcard.py implementation.  It works fine, but the performance seems low -- in my rp2 tests I see about 105ms to read 10kB of data from a file.  Whereas, the pyboads are around 2.5ms, and 16ms for the ESP32 .  I might post on the forum to discuss this - I might not have it setup properly.  This affects the ability to support wav files with a fast sample rate and stereo.  Maybe I'll look into implementing machine.SDCARD when I finish I2S for the rp2.

peterhinch, 2021-09-19:
> Does this fit with your anticipated use pattern?

I think that would simplify usage.

> wav header parser

That looks good. I was aware that the offset of 44 doesn't address all cases. I'm not sure whether to include this in the demo: the idea of these demos is to present something fairly minimal that people can build on. But I'll certainly link to it in the docs and code comments.

> sdcard.py implementation. It works fine, but the performance seems low

I encountered this a few years ago on the Pyboard 1.0 in a micropower system. To save power the Pyboard has no SD card but has an external, power switched card using that driver. The card stores a binary font file for an ePaper display. While it works fine, it's only because ePaper displays are slow. I concluded that the time to retrieve glyphs was too slow for normal displays.

As I'm sure you know, SD cards have multiple modes with different speeds. From comments in the forum they also seem to be something of a black art, with variation between manufacturers.

peterhinch, 2021-09-21:
[Here](https://github.com/peterhinch/micropython-async/blob/master/v3/as_demos/monitor/README.md) is the uasyncio monitor we discussed earlier.

The music player demo is now done. It allows you to select a directory containing wav files. Pressing "play" plays them in sequence. It has the usual play-stop-pause-replay-skip buttons. There is no facility to create playlists and I don't plan to go there - it is just a demo. Stable code has enabled me to concentrate on the I2S `play` routine, and I think there is a problem.

On the SF2 the demo runs with over 50K of free RAM. This figure is stable: there is no memory leak. Yet on occasion I see a failure allocating 25K which is my current buffer size. This occurs fairly rarely and randomly in response to an operation like moving the currency from one player button to another - an operation which uses minimal resources. The line where this is occurring is:
```python
swriter.write(wav_samples_mv[:num_read])
```
Adding debug code convinced me that the behaviour makes no sense at all. When the line runs, the `uasyncio` `out_buf` is always empty. There is no way that once in a blue moon it should allocate an entire buffer. Replacing the line with
```python
swriter.out_buf = wav_samples_mv[:num_read]
```
made no difference: there are still rare allocations taking place.

Perhaps when you've completed your work on the Pico we can work together on this one. It's too deep for me ;-)

miketeachman, 2021-09-22:
> [Here](https://github.com/peterhinch/micropython-async/blob/master/v3/as_demos/monitor/README.md) is the uasyncio monitor we discussed earlier.

Cool.  I'm going to try this out using the micro-gui example code gave me a few days ago.  At the moment the ST7735 display is wire up and showing the 3 widgets.  I just need to add the two buttons and an I2S DAC board.  Then, play with the I2S buffer sizing, and try out the monitor tool.  I hope to get a couple of hours tomorrow to continue work on this.

> Adding debug code convinced me that the behaviour makes no sense at all. When the line runs, the `uasyncio` `out_buf` is always empty. There is no way that once in a blue moon it should allocate an [...]

Wow.  I'm baffled as well.

miketeachman, 2021-09-22:
I got started on using audio.py, then hit a snag.  Here are the steps to reproduce:

- updated the library code 
- added a music folder to the SD Card
- added a wav file to the `music `folder
- added code to `hardware_setup.py` that mounts the Pyboard D SD card (`os.mount(pyb.SDCard(), "/sd")`)
- ran `audio.py` via the REPL
- pressed Select button to see what the first icon does
- got exception:

`Task exception wasn't retrieved
future: <Task> coro= <generator object 'switchcheck' at 20013d00>
Traceback (most recent call last):
  File "uasyncio/core.py", line 1, in run_until_complete
  File "gui/primitives/switch.py", line 38, in switchcheck
  File "gui/primitives/__init__.py", line 20, in launch
  File "gui/core/ugui.py", line 85, in _closure
  File "gui/core/ugui.py", line 215, in sel_ctrl
  File "gui/core/ugui.py", line 429, in do_sel
  File "gui/widgets/buttons.py", line 81, in do_sel
  File "<stdin>", line 151, in new
  File "gui/core/ugui.py", line 259, in change
  File "<stdin>", line 64, in __init__
  File "gui/widgets/listbox.py", line 40, in __init__
IndexError: list index out of range`

```
>>> os.listdir('/sd/music')
['ff-16b-2c-44100hz.wav']
>>>
```

I didn't have time to investigate.

> This is a nasty hack which improved stability

Can you describe the instability issue, so I know what to look for?

miketeachman, 2021-09-23:
I dug into the implementation to understand that '/sd/music' needs to contain other directories.  After that change it's looking good - the exception is gone (note:  it would be good to fix this behaviour).  I can see the directory after Select on the first icon, select that directory, then play the one song in the directory.

peterhinch, 2021-09-23:
Sorry for the confusion. I haven't had time to document this demo. If you only have two pushbuttons you won't be able to select albums or alter the volume. You may want to alter the volume default - [this line](https://github.com/peterhinch/micropython-micro-gui/blob/d303184b45636fc2c0eb3c5390af2c590e752ff1/gui/demos/audio.py#L84), set `value` to something nearer to 1.0. There was a bug in writer.py - make sure you have the latest version which I fixed 4 days ago. You also need short directory names for the music - another thing I'll fix.

The interesting stuff is in `play_song`.

Probably the best way to start is to strip out my debug code. [this](https://github.com/peterhinch/micropython-micro-gui/blob/d303184b45636fc2c0eb3c5390af2c590e752ff1/gui/demos/audio.py#L197) never occurs, `drain` does what it claims. [This](https://github.com/peterhinch/micropython-micro-gui/blob/d303184b45636fc2c0eb3c5390af2c590e752ff1/gui/demos/audio.py#L200) should be reinstated - the following line was a test. [This](https://github.com/peterhinch/micropython-micro-gui/blob/d303184b45636fc2c0eb3c5390af2c590e752ff1/gui/demos/audio.py#L203) should go. For reasons I don't understand it did improve stability, but you want to see the instability rather than hack it into not happening. I find that occasional allocation failures occur when moving the widget currency (pressing Next pushbutton). Sometimes the music doesn't stop cleanly when pressing Stop or Pause. The allocation failures are the clearest evidence of instability, as discussed above, because they don't make sense as Python faults.

Unfortunately the behaviour is not deterministic but I'd be surprised if you don't encounter weird issues in half an hour of use. Good luck!

miketeachman, 2021-09-23:
> Probably the best way to start is to strip out my debug code. [this](https://github.com/peterhinch/micropython-micro-gui/blob/d303184b45636fc2c0eb3c5390af2c590e752ff1/gui/demos/audio.py#L197) never [...]

I started my experimentation by reducing the I2S buffer sizes to `ibuf`=20k, and the sample buffer=10k.  This worked flawlessly for all my button tests.  I tested for around 10 minutes with this configuration.  Changing the sample buffer to 5k produced underflow and gaps in the audio stream.   Next, I made the changes you suggested.  Afterwards, it worked fine for about a minute, then I got an allocation failure for 10001 bytes.  This was unexpected as the 20s report task was consistently showing at least 50k of heap.  The steps-to-reproduce are muddled in my mind.  I had been playing with the stop/pause/play buttons not long before the allocation exception, but I don't recall the exact order.  The button actions may be unrelated to the exception.

Do you know of a way in MicroPython to report the size of the max contiguous heap space?  That would seem to be an interesting stat to report on.

Later today I'll have time to try layering in the monitor application to see if anything interesting is revealed.  I don't have a logic analyzer, but a 4-channel scope should be a good start.

Pausing a song produces the annoying buzz (expected).  On the RP2 port I fixed this by adding a `memset()` call to clear the DMA buffer when the I2S internal buffer underflows.  I'm going to make a PR for the I2S implementation on the STM32 and ESP32 ports.  This PR won't enable immediate stopping of playback, but it will still represent a significant improvement by eliminating the need to push null samples into the internal buffer.

peterhinch, 2021-09-23:
I don't know a way to report contiguous heap size. It would be very useful. However at the point where it fails I don't think it should be allocating at all. In my testing `.drain` works as expected, so `.write` is always adding to an empty buffer. To be sure of this I changed the code to
```python
swriter.out_buf = wav_samples_mv[:num_read]
```
which surely should never allocate, yet I still got rare allocation failures.

In my experience fragmentation means that repeated allocation of 10K (or in my case 25K) is bound to fail in a running system when there is only 50K available. If it was always allocating it would fail immediately. So, if it is allocating, it's only trying to do it under rare, non-deterministic, circumstances. This makes no sense. Further, you wrote the code using a memoryview so it should never allocate.

It's perhaps worth pointing out that the GUI uses nothing "clever" like timers or interrupts. Switches are polled using a task. My guess is that the allocation is a red herring. I think something is going on which is causing MicroPython to be unstable.

miketeachman, 2021-09-24:
> Adding debug code convinced me that the behaviour makes no sense at all. When the line runs, the `uasyncio` `out_buf` is always empty. There is no way that once in a blue moon it should allocate an [...]

My setup shows a different behaviour:   I added `mem_free()` debug prints before and after `swriter.write(wav_samples_mv[:num_read])`.  There is a 10k allocation every time the `Stream.write()` method is called.  10k is the sample buffer size.  

```
mem_free() b4 write() 41776
mem_free() after write() 31712

```

The allocation exception can reliably be reproduced with 2-3 iterations of these steps:
1. select album
2. play
3. pause
4. goto 2

Here's my first theory on the root cause of these allocation exceptions:  
1.  `Stream.write()` causes heap allocations, every time.
1.  The allocation exception is real.  There is not a 10k contiguous block of heap memory when  `swriter.write()` runs.

**Clues**:

1. The gc threshold always decreases by around 8k immediately before the allocation failure.  I added debugging to print the threshold before every `swriter.write()`
2. after one or two iterations of the repro steps above mem_free() **increases** after `swriter.write()` .  This suggests a GC is being done to allocate the 10k contiguous buffer (even though there is tons of free heap space -- see the debug results below).
```
threshold() b4 write() 146096
mem_free() b4 write() 36944
mem_free() after write() 49552
```

I'll put this investigation on hold until you've had a chance to check out these findings...

peterhinch, 2021-09-24:
`.write` does
```python
    def write(self, buf):
        self.out_buf += buf
```
On reflection this must allocate. The compiler doesn't know whether `.out_buf` is empty, so it must create a new object. And I can see why `.write` has to work this way, because the application might change the buffer after calling `.write`. However what it doesn't explain is why, when I replaced the call to `.write` with
```python
    swriter.out_buf = wav_samples_mv[:num_read]
```
I still got allocation failures on that line. This doesn't allocate (see below). Perhaps you might like to repeat this test. The allocation behaviour of these two approaches is confirmed by
```python
>>> import gc
>>> a = b'1234567890'*10000
>>> mv = memoryview(a)
>>> b = b''
>>> gc.collect()
>>> gc.mem_free()
87552
>>> b += mv[0:9990]  # Allocates
>>> gc.collect()
>>> gc.mem_free()
77488
>>> c = mv[0:9990]  # does not allocate
>>> gc.collect()
>>> gc.mem_free()
77504
>>> 
```

miketeachman, 2021-09-25:
> However what it doesn't explain is why, when I replaced the call to `.write` with
> 
> ```python
>     swriter.out_buf = wav_samples_mv[:num_read]
> ```

That's still a mystery for sure.

miketeachman, 2021-09-25:
When the sample stream is stopped users will hear silence rather than noise after this PR:
https://github.com/micropython/micropython/pull/7855

peterhinch, 2021-09-25:
That's a great improvement.

I doubt it's possible to play without an eventual allocation fail if it allocates the whole buffer repeatedly. So `uasyncio` needs an extra interface which does a non-allocating stream write. The user of such an interface must guarantee that the data remains unchanged until `drain` has completed. Perhaps
```python
    def write(self, buf, copy=True):
        if copy:
            self.out_buf += buf
        else:
            self.out_buf = buf  # No allocation
```
Do you agree that we should request this? If so, we should see if we can get stable behaviour with the proxy for this interface

```python
swriter.out_buf = wav_samples_mv[:num_read]
```

miketeachman, 2021-09-25:
> Do you agree that we should request this?

Definitely.  I think Python created a bit of a mess with the write/drain API.  There is some discussion that `write()` could be a co-routine and remove `drain()`. 
https://discuss.python.org/t/discussion-on-a-new-streaming-api-for-asyncio/4725/6 

I like your proposal -- adding the optional `copy `argument seems like reasonable way to make this method work within the limitations of a small memory footprint.  It keeps backwards compatibility with CPython.

peterhinch, 2021-09-26:
I was puzzled by `write()` and `drain()` from the start. Making `write` a coroutine seems so obvious there must have been a reason why this wasn't done. My _guess_ is that the devs envisaged a situation where a user would do multiple `write` calls followed by `drain`. If `write` were a coro then, in a similar scenario, if `create_task` were used then multiple `write` instances would exist. Could they have been worried about execution order? Yet there is nothing to stop a user of the existing API launching multiple `drain` instances. I remain foxed :)

I doubt the CPython devs would break code with this change. The CPython `asyncio` twists and turns over the years left me baffled - it took a while to understand exactly what `uasyncio` supported. Which is why I wrote my tutorial.

Do you want to raise a PR or shall I? I don't mind either way.

Are you getting stability writing to the buffer? I haven't managed to progress this as I'm adding SPI support to monitor - not every platform will have a spare UART. Hopefully I'll have that done today and can get back to this.

miketeachman, 2021-09-27:
> Do you want to raise a PR or shall I? I don't mind either way.

It would definitely get more attention if you submitted the PR.  I can add a supportive comment, and a Hooray reaction.

I couldn't make the audio demo fail allocation with `swriter.out_buf = wav_samples_mv[:num_read]`, using a 10k buffer.

For now, I've moved back to the rp2 I2S implementation.

peterhinch, 2021-09-27:
OK, I'll submit a PR in the next day or two. 

I'll repeat my tests with a 10K buffer. Just to be sure we're singing off the same hymn sheet, I take it your code now looks like this. I guess the last four lines can be eliminated with your PR.
```python
    async def play_song(self, song):
        wav_samples_mv = memoryview(wav_samples)
        size = len(wav_samples)
        if not self.paused:
            self.offset = 44
        swriter = self.swriter
        with open(song, "rb") as wav:
            _ = wav.seek(self.offset)
            while (num_read := wav.readinto(wav_samples_mv)) and not self.stop_play:
                I2S.shift(buf=wav_samples_mv[:num_read], bits=16, shift=self.volume)
                swriter.out_buf = wav_samples_mv[:num_read]
                await swriter.drain()
                self.offset += size
        for x in range(256):  # Necessary for silence (but not with PR?)
            wav_samples[x] = 0
        swriter.write(wav_samples_mv[:256])
        await swriter.drain()
```
I'll report back when I have a result.

peterhinch, 2021-09-28:
I have raised [this PR](https://github.com/micropython/micropython/pull/7868).

I've finished work on the monitor, which now supports SPI as an alternative to a UART for those cases where there might be no free UART available. I think this is now complete.

I'll now devote some time to the audio demo.

peterhinch, 2021-09-29:
Eureka! With your PR and the uasyncio stream tweak the audio demo is working perfectly. I need a bigger buffer to avoid occasional dropouts but is probebly due to my different display. When dropouts occur they are now quite hard to hear. Currently getting perfect audio with a 20KiB buffer and 62K of free RAM, including sustained bouts of operating the GUI. I can't break it!

I removed the code that writes 256bytes of zeros, so the PR gets hammered each time I pause playback. I'll fix the demo to check the filesystem as you suggested. Listening as I write: the core playback is now great.

The PR does seem to have transformed overall stability. Thank you for your great work on this, and your efforts in helping me fix my hardware and software issues. Please keep me posted on the Pico, I'm interested in your work on SD card throughput: I guess this applies to ESP32 too.

I guess we can close this now.

closed 2021-09-29 by peterhinch

miketeachman, 2021-09-29:
> Eureka! With your PR and the uasyncio stream tweak the audio demo is working perfectly. I need a bigger buffer to avoid occasional dropouts but is probebly due to my different display. When dropouts [...]

Wow.  Great news !   Thanks for sticking with this so we could work through the problems.  It's a really good demo to show the benefits of an asyncio design approach.  I'll put a comment on your PR right now.

peterhinch, 2021-09-29:
There are a couple of points that perhaps should be documented. I don't see these as bugs, just possible pitfalls.

1. I tried using task cancellation to stop playback. This doesn't work because the exception is thrown to `.drain()` which is evil. The code we have works fine.
2. If you interrupt the demo with ctrl-c or the code fails for any other reason, the USB link can go down. The solution is simple, a `try-finally` block as below.

```python
def test():
    try:
        Screen.change(BaseScreen)
    finally:
        audio_out.deinit()  # Necessary to ensure a clean shutdown
        print("==========  CLOSE AUDIO ==========")
```
I'm tidying up the demo but the core functionality has worked perfectly throughout. Exceptions and ctrl-c cause no problems with the above fix.