miketeachman/micropython-i2s-examples #10: Support for int. ADC via I2S ?

mtk11, 2021-11-09:
Some projects like [ESP32 Audio Input - MAX4466, MAX9814, SPH0645LM4H, INMP441](https://blog.cmgresearch.com/2020/09/12/esp32-audio-input.html) use the internal ADC via I2S for sampling. Availability of such functionality ?

miketeachman, 2021-11-12:
The `machine.I2S` class was designed to only support I2S devices.

To my knowledge there is no-one working to support analog microphones on the official MicroPython implementation.   To support an analog microphone the `machine.ADC` class could be extended, supporting devices such as the MAX4466 and the MAX9814.  At the moment, the `machine.ADC` class samples an analog signal, producing an single value;  the `machine.ADC` API can be extended, adding a method that samples the incoming analog audio signal and fills a buffer with audio samples.  e.g   `machine.ADC.readinto(buf)`.  For the ESP32 port, the ADC components of the ESP-IDF I2S peripheral interface would be used.

The Loboris port of the ESP32 allows sampling of audio data using the `machine.ADC` class.
[ADC.read_u16()](https://github.com/loboris/MicroPython_ESP32_psRAM_LoBo/wiki/adc#adcread_timeddata-freq-nsamples-1-bytetrue-datanone-waitfalse-callbacknone)

Note:  Two devices you mention, SPH0645LM4H and INPM441, are not analog devices - they are I2S devices that work with the I2S protocol and are supported by `machine.I2S`

I hope that answers your question.

miketeachman, 2022-02-21:
There are no plans to support analog microphones in the machine.I2S class, so closing this issue.

closed 2022-02-21 by miketeachman