miketeachman/micropython-i2s-examples #11: Examples broken under esp-idf V4.4

jetpax, 2021-12-15:
Thanks for the great work Mike.

I found that the significant changes in the esp-idf i2s driver since V4.3, mainly to support mclk it seems, cause some of these examples to break (eg record_mic_to_sdcard_blocking.py),  with these errors
```
E (19344) I2S: i2s_calculate_common_clock(1215): sample rate is too large
E (19344) I2S: i2s_calculate_clock(1264): Common clock calculate failed
E (19354) I2S: i2s_check_set_mclk(253): ESP32 only support to set GPIO0/GPIO1/GPIO3 as mclk signal, error GPIO number:1061222028
E (19364) I2S: i2s_set_pin(314): mclk config failed
```

I worked around this in my application, which does not use mclk by adding the following in machine_i2s.c

` i2s_config.mclk_multiple = I2S_MCLK_MULTIPLE_256;`

to  ` i2s_config`

and 

` pin_config.mck_io_num = I2S_PIN_NO_CHANGE;`

to `pin_config`

Obviously this needs a more thorough fix, especially if mclk is to be suppported...

miketeachman, 2021-12-20:
Thanks for reporting this.   I'll install esp-idf v4.4-beta1 and check this out.

miketeachman, 2021-12-22:
I did some investigation into the ESP-IDF v4.4 Beta release.  Adding the code you suggested fixes the breaking changes made in the ESP-IDF.  Thank you!

I also found two more problems:
1.  uasyncio support in I2S is broken from a bug introduced into the I2S implementation.  I submitted an Issue:
https://github.com/espressif/esp-idf/issues/8121

2.  Correct I2S operation has a dependency on the code that runs before the I2S constructor.  For example, the tone example does not run, but adding a SPI constructor ahead of the I2S constructor gets the example working -- very strange.  More debugging needed to identify the root cause of this peculiar behaviour.

ESP-IDF v4.4 is still in Beta, so I hope that Espressif will fix at least problem #1 above before the final release.

miketeachman, 2021-12-24:
I submitted a PR for this issue:
https://github.com/micropython/micropython/pull/8110

miketeachman, 2022-01-31:
The PR has been merged.

closed 2022-01-31 by miketeachman