miketeachman/micropython-i2s-examples #13: Add support for stm32 H7 series boards

mocleiri, 2022-01-11:
There are HAL differences between the STM32F4 and STM32L4 and STM32H7.

ports/stm32/machine_i2s.c includes certain .h files which eventually resolve to a header file located in the board directory like:
https://github.com/micropython/micropython/blob/d6dc4cb65a222bd05ec2746c37e457c56484e780/ports/stm32/boards/stm32f4xx_hal_conf_base.h

which points at the HAL files which are included from:
https://github.com/micropython/stm32lib/tree/1eebcda2c95d7593c68e1c81f042d23485baab26/STM32H7xx_HAL_Driver/Inc 

The HAL API for STM32H7 is different and fails to compile when MICROPY_HW_I2S1 enabled.

https://github.com/mocleiri/tensorflow-micropython-examples/issues/54

```
../../lib/stm32lib/STM32H7xx_HAL_Driver/Inc/stm32h7xx_hal_i2s.h:229:42: error: overflow in conversion from 'long unsigned int' to 'int8_t' {aka 'signed char'} changes value from '512' to '0' [-Werror=overflow]
  229 | #define I2S_DATAFORMAT_32B               (SPI_I2SCFGR_DATLEN_1)
      |                                          ^
machine_i2s.c:282:16: note: in expansion of macro 'I2S_DATAFORMAT_32B'
  282 |         return I2S_DATAFORMAT_32B;
      |                ^~~~~~~~~~~~~~~~~~
machine_i2s.c: In function 'i2s_init':
machine_i2s.c:583:9: warning: implicit declaration of function '__HAL_RCC_PLLI2S_ENABLE' [-Wimplicit-function-declaration]
  583 |         __HAL_RCC_PLLI2S_ENABLE();  // start I2S clock
      |         ^~~~~~~~~~~~~~~~~~~~~~~
machine_i2s.c: In function 'machine_i2s_init_helper':
machine_i2s.c:793:9: error: 'I2S_InitTypeDef' has no member named 'ClockSource'
  793 |     init->ClockSource = I2S_CLOCK_PLL;
      |         ^~
machine_i2s.c:793:25: error: 'I2S_CLOCK_PLL' undeclared (first use in this function); did you mean 'IS_RCC_PLL'?
  793 |     init->ClockSource = I2S_CLOCK_PLL;
      |                         ^~~~~~~~~~~~~
      |                         IS_RCC_PLL
machine_i2s.c:793:25: note: each undeclared identifier is reported only once for each function it appears in
cc1: all warnings being treated as errors
```

I think you should be able to use the $(MCU_SERIES) make file variable to use the h7 HAL when building an H7 board versus the standard HAL when building an f4 or l4.

I'm not very familiar with stm32 so I thought I would file this issue to see if its something that looks straight forward to you.

If its more complicated I'm happy to take a stab at fixing it and file a pull request if I can solve it.

miketeachman, 2022-01-11:
Unfortunately, I don't see an easy fix.  The compiler errors point to differences in I2S implementation on the H7 processor series.  But, it's likely possible to adapt machine_i2s.c and dma.c (possibly others ...) to accommodate these differences.

Here's a possible path forward:
- create a new branch based on micropython master
- implement H7 support on this new branch, using pre-processor macros as needed, e.g.  #if defined(STM32H7), etc
- I can help with regression testing for the F4 and F7 processors (what is supported now).  Using your branch I'll test against my hardware (pyboard v1.1 and pyboard D)
- submit a PR to the micropython project.  I'll add a comment supporting the PR, indicating that I've done regression against the pyboard hardware.

How does that sound?

mocleiri, 2022-01-11:
Thanks for taking a look.  The path you suggest sounds good to me.  

I will close this issue and open a new one when I have an H7 implementation for you to regression test.

closed 2022-01-11 by mocleiri