miketeachman/micropython-i2s-examples #19: Sd library

medvosa, 2022-12-05:
Where did you get sd library, my pi pico has not it as module nor as machine.sdcard.
Thanks in advance

miketeachman, 2022-12-05:
I recommend using the `sdcard.py` driver in the `micropython-lib` repo:
https://github.com/micropython/micropython-lib/blob/master/micropython/drivers/storage/sdcard/sdcard.py

tgmaxx, 2022-12-31:
I am trying the examples on a Raspberry Pi Pico. I can run "play_tone.py" with no problems. However, I am having problems with the SD card driver for the other examples. First. the link to sdcard.py on the page https://github.com/miketeachman/micropython-i2s-examples is broken. However, I found the sdcard.py file in the repo at the new location as you mention above. When I try the example "play_wav_from_sdcard_blocking.py" it comes up with errors. Specifically, 

Traceback (most recent call last):
  File "<stdin>", line 76, in <module>
  File "/lib/sdcard.py", line 259, in readblocks
  File "/lib/sdcard.py", line 197, in readinto
OSError: timeout waiting for response

miketeachman, 2023-01-06:
Thanks for reporting this broken link and describing the problem you found with this SD card module.  I'll fix the broken link right now.

closed 2023-03-03 by miketeachman