miketeachman/micropython-i2s-examples #21: Shift before writing to speaker

sammachin, 2023-03-25:
I'm receiving PCM audio frames via a network interface and then in the callback for that I call I2S.write.

Everything works well apart from the audio is a little quiet, especially when compared to a locally generated tone.

I've tried to use I2S.shift on the data before i call write but I get the error:
`TypeError: object with buffer protocol required`

this is my code, the audio data is in msg. and spk is my I2S device
```
 if ttype == 'aud':
            I2S.shift(buf=msg, bits=16, shift=2)
            spk.write(msg)
```

miketeachman, 2023-03-26:
Can you share more of the code that shows this problem?

miketeachman, 2023-05-13:
No response to the request for code showing the problem, so I assume the issue has been resolved.

closed 2023-05-13 by miketeachman