miketeachman/micropython-i2s-examples #23: Support PCM5100A ?

jinyistudio534, 2023-05-13:
Can i use this library with PCM5100A ?

miketeachman, 2023-05-13:
Yes.  The library will work with a PCM5100A device.

jinyistudio534, 2023-05-21:
Do you think this library can control pico-audio? Do I need to change those codes?
https://shop.pimoroni.com/products/pico-audio-pack?variant=32369490853971

miketeachman, 2023-05-21:
Yes.  The pico-audio can be controlled with MicroPython, using pins 12
(SD), 14(SCK), and 15(WS).  Here is a simple example
<https://github.com/miketeachman/micropython-i2s-examples/blob/master/examples/play_tone.py>
to try.  You will need to change the code to use pins 12, 14, and 15.

miketeachman, 2023-05-21:
Yes.  The pico-audio can be controlled with MicroPython, using pins 12 (SD), 14(SCK), and 15(WS).  Here is a [simple example](https://github.com/miketeachman/micropython-i2s-examples/blob/master/examples/play_tone.py) to try.  You will need to change the code to use pins 12, 14, and 15 as shown below :

```
elif os.uname().machine.count("Raspberry"):

    # ======= I2S CONFIGURATION =======
    SCK_PIN = 14
    WS_PIN = 15
    SD_PIN = 12
    I2S_ID = 0
    BUFFER_LENGTH_IN_BYTES = 2000
    # ======= I2S CONFIGURATION =======`
```

jinyistudio534, 2023-05-22:
12=GP9,14=GP10,15=GP11 👍
elif os.uname().machine.count("Raspberry"):
    print("Raspberry")
    # ======= I2S CONFIGURATION =======
    SCK_PIN = 10
    WS_PIN = 11
    SD_PIN = 9
    I2S_ID = 0
    BUFFER_LENGTH_IN_BYTES = 2000

miketeachman, 2023-05-22:
Great !   Thanks for catching my mistake.  I hope you did not waste too much time to find my mistake.