miketeachman/micropython-i2s-examples #33: Raspberry pi pico - record to sd card: error

Hubert662, 2024-08-11:
Hello,
First of all, thank you for your post. Great job.
I have a raspberry pi pico and I want to record audio to an SD card. I used your "record_mic_to_sdcard_non_blocking.py" file, and didin't change anything. I copied the "sdcard.py" file to the pico, and again didin't change anything. I have a microphone INMP441 and micro SD card module like this:
[image]
Everything together:
[image]
But something is wrong, and this happend:
[image]
What could be wrong? What am I doing wrong?
I'm a beginner so maybe I made some simple mistake.

miketeachman, 2024-08-22:
Sorry for the delay to respond.  Do you still have a problem?
If so, please provide the entire micropython code listing.  It is difficult to help debug the problem with only a few lines of code.  Thanks!

Hubert662, 2024-08-22:
Thanks for your reply. I solved this problem. I changed "SPI bus speed to SD card" to 20.000.000. Now it works but I have different problem after closing file. I will write about it on the second issue comment.

closed 2024-08-22 by Hubert662