miketeachman/micropython-i2s-examples #35: [Question] Does this work with mono magnet speakers?

aimannajjar, 2024-10-18:
Hello,
Sorry for the beginner question, I was able to get this library to wotk on my pico + PCM5102A board, but it only works with headphones. 

I tried to hook it with a magnet speaker using a terminal like this: https://a.co/d/744VIPg but there was no sound output. For reference, my speaker is this: https://a.co/d/amTXBMY and it works if I output audio using PWM ports, but doesn't seem to work with I2S.

Also note I tried connecting both R and L channels (obe at a time) but neither worked.

I'd appreciate any pointers. Thanks!

miketeachman, 2024-10-21:
Hi !
Unfortunately the PCM5102A IC does not have enough output power to drive speakers.  However, as you found, it has enough power to drive headphones.  To drive speakers you would need to add some sort of amplifier module between the output and the speakers.   To drive speakers I use the I2S 3W amplifier, MAX98357A.  Here is an example breakout board from Adafruit:  https://www.adafruit.com/product/3006

aimannajjar, 2024-10-21:
Hi @miketeachman , thanks so much for the reply! That is exactly what I ended up doing and it worked flawlessly, thanks again for your help! :)