miketeachman/micropython-i2s-examples #40: OSError [Error 22] EINVAL line 133 wavplayer.py

p26grinz, 2025-01-31:
I'm using easy_wav_player as driver. Pico 2 with micropython V1.24.1.uf2.  I download this repo Jan 12, 2025.
The os.listdir(self.root) doesn't like the trailing '/' in self.root resulting in the title message.
I commented out the trailing '/' in line 41 and the error went away.
Updated line 140 insert "/" as shown:            self.wav = open(self.root + "/" + wav_file, "rb")
and I could now here the .WAV files, yea!

Thanks for developing these examples.  Morris

miketeachman, 2025-02-01:
Hi!
Thanks for taking the time to report this error.  I don't yet have a Pico 2 so I tested the `wavplayer.py` example with a Pico using the same firmware release, V1.24.1.uf2.  The example worked on the Pico.  It looks like this problem is unique to the new Pico 2.  I'll have to put in an order for a Pico 2 to debug this further.  I've added a Pico 2 to my DigiKey cart !

miketeachman, 2025-02-13:
My Pico 2 board arrived and I tested the example with firmware release, V1.24.1.uf2.

The example ran without error.

I'm a bit baffled on the issue you are seeing.  I wonder what is different between our environments.  Can I ask you to run the minimal test shown below (after substituting the GPIO pins that you use for the SD card module) as see what happens... thanks

```
import os
from machine import Pin
from sdcard import SDCard
from machine import SPI

cs = Pin(13, machine.Pin.OUT)
spi = SPI(
	1,
	baudrate=1_000_000,  # this has no effect on spi bus speed to SD Card
	polarity=0,
	phase=0,
	bits=8,
	firstbit=machine.SPI.MSB,
	sck=Pin(14),
	mosi=Pin(15),
	miso=Pin(12),
)

sd = SDCard(spi, cs)
sd.init_spi(25_000_000)  # increase SPI bus speed to SD card
os.mount(sd, "/sd")
os.listdir("/sd")
os.listdir("/sd/")
```